lsochanowski/GoHeishaMon #4: When writing enable?

Dario82, 2020-05-17:
Hello,
I've realized a golang version that uses cloud for read/write on mqtt and I can switch on / off and other functions from hassio. My solution is a wrapper with services cloud (for get data) and smart cloud (for set data). Your project is very intersting because run on heatpump directly.

When writing enable?

Thank you,
Dario.

MiG-41, 2020-05-17:
Hi , what do you mean by "writing enable" ?

Dario82, 2020-05-17:
this:
`if config.Readonly != true {
		log_message("Not sending this command. Heishamon in listen only mode! - this POC version don't support writing yet....")
		os.Exit(0)
	}`

MiG-41, 2020-05-17:
This is some restovers ( or future plans) from @lsochanowski ...
But don't worry with this function, Set commands works , and in the same way like in orginal HeishaMon project : https://github.com/Egyras/HeishaMon/blob/master/MQTT-Topics.md

Dario82, 2020-05-17:
Works? ok, I'll try!
Thank you!

closed 2020-05-17 by Dario82

lsochanowski, 2020-05-17:
Main part of Goheishamon is ported directly from heishamon project this is why this feature is Here even if for now it’s make No sense i’ve decided to leave it in the code because of plan and poc to use goheishamon concurent with a2wmain