lsochanowski/GoHeishaMon #9: No luci onsystem

Dario82, 2020-05-22:


MiG-41, 2020-05-22:
In web browser you should put ip address of CZ-TAW1 ,and luci should apear.

Dario82, 2020-05-22:
I do it!
I have SSH only... MQTT ok.

MiG-41, 2020-05-22:
Please try 1.0.134 realease , this one tested. Don't put dropbear file to drive (ssh is already included).
To test i ,go back first to SmartCloud ( simplest way to press three buttons ,when GoHeisha works and the other "side" with Smart Cloud strart). And then put 1.0.134 realease to drive ( optionally with config file). And  look.

Dario82, 2020-05-22:
Only (release v. 1.0.134):
- openwrt-ar71xx-generic-cus531-16M-kernel.bin
- openwrt-ar71xx-generic-cus531-16M-rootfs-squashfs.bin
- GoHeishaMonConfig.new

on the drive?

MiG-41, 2020-05-22:
yes.
BTW , you are not loosing much , becouse in luci you can only set a wireless comunication max , but it is courious ,why you don't have it...

Dario82, 2020-05-22:
Must I add a "EnableCommand=true" on config?

MiG-41, 2020-05-22:
you can . Probably it is not a must , but i'm not sure.

Dario82, 2020-05-22:
I've pressed 3 button on the same time... nothing... How I can return to original firmware?

MiG-41, 2020-05-22:
GoHeishaMon should be active when you press buttons. ( color of diode will be changed to veryfy it ).
You can do it manually in ssh giving commands:
fwupdate sw 
reboot

Dario82, 2020-05-22:
Now... I don't have SSH, MQTT, WEB...

MiG-41, 2020-05-22:
you should wait some minutes for ssh. 
Is the middle LED have the white color ?

Dario82, 2020-05-22:
The middle led is white. I can only ping my device.
But, how this device can connect to my wlan without password? It uses the configuration in other side of original firmware?

Dario82, 2020-05-22:
ssh: connect to host 192.168.10.17 port 22: Connection refused
...nothing...

MiG-41, 2020-05-22:
Yes , the settings are generally identical for booth sides.
If LED is white ,then something is wrong with config file and conigurations ( GoHeishaMon starts , and stops).

Dario82, 2020-05-22:
I've restarted the pump, on / off.
Now I have MQTT (data sent ok) but I don't have luci and ssh.
But I see a ESPxxxx Wifi, is it normal?

MiG-41, 2020-05-22:
ESPxxx WiFi is not normal , but i doupt is comming from CZ-TAW1. Maybe you have somwhere such device ESP8266 or ESP32 ,with is additionally sending it ? 
In my small project i had same , used ESP8266 ,connecting to my router , but in the same time ESP8266 generate some ESPxxxnetwork , like here https://forum.arduino.cc/index.php?topic=501263.0

This is strange ,why ssh and web i have , and you don't....

MiG-41, 2020-05-22:
When you have "EnableCommand=true" , can you listen topic in mqtt "panasonic_heat_pump/OSCommand" ( i.e. via mosquitto_sub) , and then send to this topic command like "uptime" or "ls /tmp" to see if it works ?

MiG-41, 2020-05-22:
Can you join to Slack communication , given in Readme.md in Heishamon project : https://github.com/Egyras/HeishaMon/tree/master/HeishaMon
would be easier to discuss

lsochanowski, 2020-06-16:
i think it was resolved in newest version ssh  will work without configuring over mqtt

closed 2020-06-16 by lsochanowski