lsochanowski/GoHeishaMon #11: implementing some kind of uboot boot script

lsochanowski, 2020-06-16:
sugested by stefanbb :

would it be possible to implement some kind of uboot boot script or similar in further versions? That would make handling with ssh and network settings much more easier, I think. And we could avoid the fiddly handling during soldering these small cables. What do you think? Is that a possible improvement? It would help some stupid beginners like me a lot in setting up and administration. And if I make an error, I just would have to change the boot script.

MiG-41, 2020-06-19:
@stefanbbb , can you describe more , what this menu should present , what it does ? ,and where to display ?

stefanbbb, 2020-06-21:
Some time ago, I had some projects with uboot on Raspberry. Here  I figured out, that there is an option to use a so called boot script. This is a compiled textfile with the option to configure most of the important OS parameter like network, filessystem and so on (please see this link: https://newit.co.uk/forum/index.php?topic=2743.0). The boot script will be read by uboot during startup. The CZ-TAW1 seems also to use uboot. Therefore it came to my mind, that it would be a great possibility to setup all important system parameters without any console, just by changing the boot script.