lsochanowski/GoHeishaMon #14: Any other debugging options?

Uriziel01, 2021-01-14:
Hi,
I'm facing an issue that my HomeAssistant mqtt broker gets the messages from CZ-TAW1 but HAAutoDiscovery does not work properly, only value messages are posted to mqtt, no configuration messages at all. I also cannot do any "write" commands, I only get values from the board, actions like below does nothing.

[image]

Are there any additional ways except soldering serial port wires to the board to see what is happening? Maybe there should be a toggleable (most propably another configuration field) debug mqtt topic that posts console output?

My config:
```
Readonly=false
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=5
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID="[REMOVED]"
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.[REMOVED]"
MqttPort="1883"
MqttLogin="[REMOVED]"
MqttPass="[REMOVED]"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true

```

P.S - Dzięki za Twoją ciężką pracę, super projekt!

Uriziel01, 2021-01-14:
Maybe even something simple and basic like:
```go
func logHex(command []byte, length int, mclient mqtt.Client) {
	fmt.Printf("% X \n", string(command))
	if config.Logmqtt == true {
		logHex(command, len(command))
                mclient.Publish(fmt.Sprintf("%s/mqttlog", config.Mqtt_set_base), byte(0), false, command)
	}
}
```

lsochanowski, 2021-01-14:
Hi,

In our taw image there is SSH you can login to taw and run goheisha in interactive mode to watch detailed log ,also there is topic to execute commands on taw by mqtt Messages 

Ha autodiscover is not finished yet ;/ and it's expected behavior you need to set topics manualy 

There is two thing curently ongoing in project autodiscover and optionboard implementation

Uriziel01, 2021-01-14:
The mqtt OSCommand topic is also not working for me sadly so I'm unable to enable ssh that way :/ Any precompiled image with SSH enabled available?

I could help with HA AutoDiscovery (also want to have it working on my instance) if you are opened for PRs?

lsochanowski, 2021-01-14:
PR's are welcome ! I'am stuck with that because i have to decide what type of entity is best for a write command (mqtt thermostat or input etc ) and because of this i stoped working with that feature ;)

lsochanowski, 2021-01-14:
Every image is with SSH

lsochanowski, 2021-01-14:
Oscommand must work can you give me some more info ?

MiG-41, 2021-01-14:
SET functions are implemented only up to SET13 , so SetFloorHeatDelta not working (yet...)

MiG-41, 2021-01-14:
> Are there any additional ways except soldering serial port wires to the board to see what is happening?

Also soltering USB serial conwerter not help ,as i remember , since after starting GoHeishaMon serial console is blocking ( probabl it is some issue with serial speed  , this one works with 115200 , GoHeishaMon change it to 9600 ( just like communication with HP) ,and no more logging...
But ssh is the best way.

MiG-41, 2021-01-14:
And finally for Home Assistant there is ready some example on HeishaMon repository , and waiting for working HAAutoDiscovery you can always use: https://github.com/Egyras/HeishaMon/tree/master/Integrations/Home%20Assistant

lsochanowski, 2021-01-14:
[image]
i've checked and command work like designed ;)

Uriziel01, 2021-01-14:
I've tried that and the result is sadly different :/ I'm sure this is some really dumb misteake on my configuration(?) side, but just cannot pinpoint what it is.

[image]

[image]

Uriziel01, 2021-01-14:
It's propably not be relevant at all but I'm not using WiFi, connecting only using ethernet (my wifi signal is very weak in the pump room)

Uriziel01, 2021-01-14:
@MiG-41 Thanks for the HASS yaml, it's way better that way than defining all of them on my own, had no idea there is such file. Tried that but it is really tedious 😋

Uriziel01, 2021-01-14:
You guys are right regarding the SET13 thing, tried holiday mode that is supported and it works, it enabled it on the pump. Still no luck with OSCommand

lsochanowski, 2021-01-14:
ok what command you are listen to ? ;)

lsochanowski, 2021-01-14:
ok i've seen  you'r problem ;) i think so ;) 

in you first post you give your config and you have 

Mqtt_set_base="panasonic_heat_pump/commands"

then. OSCommand shoult'd be 
panasonic_heat_pump/commands/OSCommand
and listen for out 
panasonic_heat_pump/commands/OSCommand/out

it's because :
c.Subscribe(config.Mqtt_set_base+"/OSCommand", 2, HandleOSCommand)

Uriziel01, 2021-01-14:
@lsochanowski Jesus... you are so right 👍 Now it's working as intended 🙇

MiG-41, 2021-01-15:
So SSH is not working at your  case ? Did you tried already Firstboot method described in read.me ? If not , can you log the command "uci show" via OSCommand/out and pass it ? ( there can be wifi creditials to remove)

Uriziel01, 2021-01-15:
@MiG-41 The hint from @lsochanowski resolved my issue with SSH ✌️ I've managed to compile my own modified `taw2mqtt.go` and I hope I'll be able to address some issues with PRs soon.

lsochanowski, 2021-01-15:
Did you manage to enable SSH by mqtt can you provide some more info because we want to make sure it's covered by readme and our test taw are modified we need some Feedback from someone who do it on ,,clear" taw like you ;)

Uriziel01, 2021-01-15:
Sure, I'll try to also improve README content to account for my issue. Hope will have some time to do so in the upcomming weekend.

MiG-41, 2021-01-15:
@Uriziel01 ,but you can also tell us here , what have to done additionally ?
Aslo what are your's modifications ? Maybe it can be siutable for more people  :)

MiG-41, 2021-02-06:
@Uriziel01 any news ?