lsochanowski/GoHeishaMon #15: Cannot connect after flashing

belowm, 2021-02-07:
Hey, first of all, thanks creating this project!
I followed the installation process and observed the LED changes as described. I unplugged the pen driver after the white light turned off. 
Now, power is green, status is white, and link is green (blinking). Does this pattern indicate normal operation? 
Is wifi configuration still done via WPS? 
I tried connecting via LAN, but I don't get any IP assigned via DHCP. Assigning a static IP in the 192.168.1.x range didn't help either. Any hints are much appreciated. 

Thanks!

MiG-41, 2021-02-07:
Led status sounds ok.

Regarding the connection , we assumed , that conection will be identical ,to one with was set during "normal" (Smart Cloud) setting.  
So no ,( in current version) , button WPS dont work ,when you run on GoHeishamon side.
Please set your destination conection first ,and then start GoHeishamon procedure. 
If you need change the connection ,the easiest way will be to go back to Service Cloud ( by pressing 3 buttons ) , and set destination conection there.

belowm, 2021-02-08:
Thanks, I'll give this a try. I didn't imagine that wifi configuration is kept after flashing the firmware.

MiG-41, 2021-02-08:
Problem is , that we very seldom receive a feedback.... In our cases works , and assuming that for others too :) 
Give feedback !  :)

belowm, 2021-02-10:
Works great! Thanks man. I think the readme should include more detailed instructions for setting up. Maybe I can contribute? Closing this for now.

closed 2021-02-10 by belowm

MiG-41, 2021-02-10:
@belowm , if you have something to add , you are welcome to made a PR.
Thank you in advance !