lsochanowski/GoHeishaMon #16: Home Assistant MQTT Discovery not working

skortela, 2021-03-13:
Hi, I installed latest version (1.0.166) to CZ-TAW1. MQTT is working, i get can see mqtt messages with proper data.
However it didn't send any home assistant MQTT discovery configurations.
Should it work, or did I miss something?

My config:

Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID=""
Mqtt_topic_base="vilp/main"
Mqtt_set_base="vilp/commands"
ForceRefreshTime=300
MqttServer="192.168.1.5"
MqttPort="1883"
MqttLogin="xxxx"
MqttPass="xxxx"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true

MiG-41, 2021-03-13:
If you really need this feature try this fork : https://github.com/rondoval/GoHeishaMon
It will be also good  ,that rondoval will make a Pull Request , since his modificatios are really great.

skortela, 2021-03-14:
Ok, thanks.

closed 2021-03-14 by skortela