lsochanowski/GoHeishaMon #17: Connection lost after router reboot

cigas4, 2021-09-03:
Everything was working fine until I had to restart my router. After that connection to GoHeishaMon was lost and never recovered (all entities went to unavailable in Homeassistant).
I am using CZ-TAW1 module connected over eth cable.
It is not retrying to reconnect after connection was lost?
How should I proceed in this situation? I don’t want to power cycle complete A2W system every time I reboot my router.

lsochanowski, 2021-09-03:
did you enable ssh access? on taw? 
if so can you send me a logs from goheisha? i cannot reproduce ;/

cigas4, 2021-09-04:
> did you enable ssh access? on taw?
> if so can you send me a logs from goheisha? i cannot reproduce ;/

Hi, I do have “EnableCommand=true” in my config file but I am not able to connect to ssh.

lsochanowski, 2021-09-04:
but there is more to do than only seting enableconfig  to enable ssh access 
please look at readme
https://github.com/lsochanowski/GoHeishaMon

cigas4, 2021-09-06:
> 
> 
> but there is more to do than only seting enableconfig to enable ssh access
> please look at readme
> https://github.com/lsochanowski/GoHeishaMon

Hi, yes you are right, I tried executing these following steps, but it looks like nothing happened:

`Public in MQTT topic "panasonic_heat_pump/OSCommand" (or eqvivalent with is set as Mqtt_set_base) one by one values: "umount /overlay" , "jffs2reset -y" and finally "reboot". This will perform a so called firstboot. You can see the output console in topic"panasonic_heat_pump/OSCommand/out".`

From HA I sent MQTT message to **_panasonic_heat_pump/OSCommand_** with payload **_umount /overlay_**, **_jffs2reset -y_** and **_reboot_**. But nothing happened, also I saw no messages coming on topic **_panasonic_heat_pump/OSCommand/out_**.
What is correct way to do it? Maybe topic is not correct, what do you mean _**"or eqvivalent with is set as Mqtt_set_base"**_?

I have also tried restarting my router one again, then also HA itself, but issue didn't appeared this time. Last time when issue appeared, I shut down HA and modem/router for some electricity works (for about an hour or so). CZ was on during this time. Maybe longer time without connection LAN or MQTT server causes issue.