lsochanowski/GoHeishaMon #21: Curves settings.

pawelkli, 2022-04-14:
Hi,
I am trying to change the target temperature in TOP29-main/Z1_Heat_Curve_Target_High_Temp. According to the information from HeishaMon, I should use a JSON document.
I tried to change with the HP on and with the HP off. There is no response from the pump. Is it possible that all functionality from HeishaMon is not supported in this version?

mosquitto_pub -h 192.168.X.X -p XXXX -t "panasonic_heat_pump/commands/SetCurves" -m '{"zone1":{"heat":{"target":{"high":45,"low":25}}}}' -u [user] -P [pass]

I was able to change other parameters.
GoHeishaMon-1.0.166

Regards, Great Job

MiG-41, 2022-04-14:
At my place your  command works (sets panasonic_heat_pump/main/Z1_Heat_Curve_Target_High_Temp to 45 and panasonic_heat_pump/main/Z1_Heat_Curve_Target_Low_Temp to 25).

pawelkli, 2022-04-14:
Ok, thank You for the check.

closed 2022-04-14 by pawelkli