lsochanowski/GoHeishaMon #22: Problem connecting through Ethernet after flashing

robin171, 2022-05-04:
Hi,

I hope you help me with getting my CZ-TAW1 up and running with Heishamon. I cannot connect to it via MQTT. What I have done:

1. Connected CZ-TAW1 to Aquarea Smart Cloud via LAN/Ethernet --> working
2. Downloaded .bin files of version 1.0.166 of the default branch
3. Copied the files to an USB stick with an additional txt file "GoHeishaMonConfig.new" with the following contents:

> Readonly=true
> Device="/dev/ttyUSB0"
> Loghex=true
> ReadInterval=1
> Aquarea2mqttCompatible=false
> Aquarea2mqttPumpID=""
> Mqtt_topic_base="panasonic_heat_pump/main"
> Mqtt_set_base="panasonic_heat_pump/commands"
> ForceRefreshTime=300
> MqttServer="127.0.0.1"
> MqttPort="1883"
> MqttLogin=""
> MqttPass=""
> SleepAfterCommand=1
> MqttClientID="GoHeishaMon-pub"
> MqttKeepalive=60
> EnableCommand=true
> HAAutoDiscover=true

4. inserted the USB drive into the CZ-TAW1 and pressed the 3 buttons for 10 seconds until the middle LED starts flashing green / red / blue
5. Waited until the white LED was on and off again and removed the USB drive 

It seems the proces of installing new firmware was successful, but unfortunately I cannot connect to MQTT. The top LED is green, middle white and bottom blinking green. In the DHCP table of my router I can see an IP adress is assigned to the module. I have tried a double reset, but also no success. I try to connect with Openhab3 and I have tried a MQTT test tool for Windows. I use no encryption, port 1883.

I'm lost, can anyone help me?

Thanks,

Robin

MiG-41, 2022-05-04:
You have wrong IP adres of MQTT server , for sure it is 127.0.0.1 ( this adres would be correct , if MQTT server would be on CZ-TAW1 together with GoHeishamon , but there is not).

robin171, 2022-05-05:
Thanks for your quick reply! 

I was thinking the wrong way around, the CZ-TAW1 being the server. It works now!

closed 2022-05-05 by robin171