lsochanowski/GoHeishaMon #23: Can't access SSH

alberto072710, 2022-06-12:
Hello,

First of all thank you very much for your work. I have successfully installed the software. But when trying to enable ssh and luci by sending the commands through mqtt, the device does not connect with mqtt or with ssh. I have sent the commands through mqtt "umount /overlay" , "jffs2reset -y"and "reboot" with the outputs that I attach in the image. I need to configure access by WIFI. What I can be doing wrong
[image]
Thanks!

Human3322, 2022-09-07:
Thanks you very much for your fantastic work. I also have the same problem, can't find SSH in the luci web interface.
All instructions in the "readme" works fine and I'm not advanced user so what should I do to see the last picture with walues from heat pump?
Regards

burt2902, 2022-11-30:
There is a way to access ssh without going thru first-boot procedure. Assuming MQTT is working, the steps require sending MQTT OS command to spawn ssh server and using deprecated cipher algorithm when ssh-ing.
```
mosquitto_pub -t "panasonic_heat_pump/commands/OSCommand" -m "/usr/sbin/dropbear"
ssh -l root -oKexAlgorithms=+diffie-hellman-group1-sha1 <GOHEISHAMON IP>
```

pando85, 2023-10-16:
Thanks, it worked perfectly. After it, I fixed the dropbear config with this:
```bash
# cat /etc/config/dropbear
config dropbear
    option PasswordAuth 'on'
    option RootPasswordAuth 'on'
    option Port            '22'
#    option BannerFile    '/etc/banner'
```
And after restarted it worked.

adnovea, 2023-11-05:
Thanks Pando85 for you config, it works like a charm.

**GoHeisaMon on CZ-TAW1B**
I had to struggle to have a version installed and running on my CZ-TAW1B module (see [link](https://github.com/lsochanowski/GoHeishaMon/issues/24#issuecomment-1701207096))

**Enabling SSH**
As a beginner to HA and not a software engineer, I spent lot of time understanding where to start from. 
Especially regarding the MQTT support.
To issue the MQTT command for SSH, I used the **MQTT Explorer** window application but I should have used the **MQTT integration** (_Configure_ button) to send the command:
```
    topic :     panasonic_heat_pump/commands/OSCommand
    Payload :   /usr/sbin/dropbear
```

**Make SSH permanent**
Then to create the /etc/config/dropbear I was blocked. I found that a console like PUTTY is very helpful to create text file.
To copy console to file just type the following command:
```
  cat > /etc/config/dropbear
```
then copy-paste the content of the pando85 post below and save with **CTRL+D** .
```
config dropbear
    option PasswordAuth 'on'
    option RootPasswordAuth 'on'
    option Port            '22'
```
To check that everything is correct type the _cat /etc/config/dropbear_ command.

**Files transfert**
The good news is that **WinSCP**  is supported and will help you to copy file to GoHeisaMon.
You have to select the SCP Protocol and connect to the module.

**OPGK and Free space**
I discovered that OPJKG could not be updated due to an obsolete repository path.
I tried to update the repository path (_LuCi > System > Software > Configuration_) with the following line :
```
src/gz ioe_device http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/base
```
I was able to update (_opkg update_) but no new modules could be installed because there is not enough space left (about 237kB) on the CZ-TAW1B.

**Conclusions**
SSH and WinSCP are very helpful to debug and add script to work-around the crashes (_see my other post_)

stapad, 2024-02-05:
That is old topic but I do not want to create another one as my question is related with ssh (and also lua) - I'm asked to provide root password which I haven't got setup in a first place. What is the initial root password so that I can login for a fist time and change it?