lsochanowski/GoHeishaMon #24: Can't install

sabatpoznan, 2022-09-23:
Hi,

I have files listed below on pendrive (FAT32 - tied 6 different):

openwrt-ar71xx-generic-cus531-16M-rootfs-squashfs.bin
openwrt-ar71xx-generic-cus531-16M-kernel.bin
GoHeishaMonConfig.new 

`eadonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID="B0***-*****"
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="1**.***.**.***"
MqttPort="1883"
MqttLogin="******"
MqttPass="******"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true`

After 10-15s of pressing 3 buttons on TAW-CZ1 and middle led is changing colours, after a while it goes beak to blue and than starts flashing.

What am I doing wrong?

adnovea, 2023-08-31:
Hi,
**Installation hangs with center LED steady white**

Everything goes well until **CA-TAW1B** reboot and the center LED lights white for ever.
After 5 mins, I removed the USB stick.
I can ping and connect the module using LuCI but neither MQTT nor SSH works.
It seems there is a configuration file issue, so I converted the **GoHeishaMonConfig.new** text file into UTF-8 just in case but w/o change.

I run Home Assistant on a RPi at IP:192.168.0.120 with Mosquitto broker and MQTT integration module.
My config is : 
```
Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID="XXXXX-XXXXX"   <-- Is the module ID required because sometimes I read that it should be left blank
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.0.120"
MqttPort="1883"
MqttLogin="xxxxxxx"
MqttPass="xxxxxxxx"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true
```

Thanks for your help

adnovea, 2023-09-01:
[SOLVED]

I fixed my problem using the post https://github.com/lsochanowski/GoHeishaMon/issues/27#issuecomment-1374619385
My **CZ-TAW1B** runs with the GoHeishaMon **1.1.191z166** version but not with the 1.0.166.

NOTE: 
_After depressing the 3 buttons more than 10 secs, the update starts and takes about 3 mins then the module reboot.
When the leds are as described below, one can remove the USB stick._
  POWER LED : steady green
  STATUS LED : steady white 
  LINK LED : slowly blink

Many thanks to **lsochanowski** for the great job achieved.