lsochanowski/GoHeishaMon #26: GoHeishaMon disconnects from MQTT brocker on CONNACK

burt2902, 2022-11-28:
Can't get GoHeishaMon to publish messages to Mosquitto brocker.

GoHeishaMon 1.0.166 successfully installed on CZ-TAW1 with the following changes to the default config:
- MqttServer="192.168.1.2"
- MqttPort="1883"
- MqttLogin="aquarea"
- MqttPass="REMOVED"

Mosquitto server running on 192.168.1.2 receives messages from other clients and receives only one connection attempt from GoHeishaMon-Pub client (without messages) after each CZ-TAW1 reboot.

Mosquitto verboe logs:
1669647699: New connection from 192.168.1.196:52870 on port 1883.
1669647699: New client connected from 192.168.1.196:52870 as GoHeishaMon-pub (p2, c1, k60, u'aquarea').
1669647699: Will message specified (7 bytes) (r1, q1).
1669647699: 	panasonic_heat_pump/commands/LWT
1669647699: Sending CONNACK to GoHeishaMon-pub (0, 0)
1669647699: Client GoHeishaMon-pub closed its connection.

GoHeishaMon does not respond to messages so enabling ssh is impossible. Luci interface works.

Any ideas?

burt2902, 2022-11-28:
Looks like LuCi system log indicates a crash...

```
Mon Nov 28 00:51:54 2022 user.emerg syslog: panic: runtime error: invalid memory address or nil pointer dereference
Mon Nov 28 00:51:54 2022 user.emerg syslog: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x303c3c]
Mon Nov 28 00:51:54 2022 user.emerg syslog: 
Mon Nov 28 00:51:54 2022 user.emerg syslog: goroutine 1 [running]:
Mon Nov 28 00:51:54 2022 user.emerg syslog: go.bug.st/serial.(*unixPort).Write(0x0, 0x5d4a40, 0x6e, 0x6e, 0x1223ac00, 0x8b, 0xb2c97000)
Mon Nov 28 00:51:54 2022 user.emerg syslog: 	/go/src/go.bug.st/serial/serial_unix.go:88 +0x30
Mon Nov 28 00:51:54 2022 user.emerg syslog: main.send_command(0x5d4a40, 0x6e, 0x6e, 0x6e, 0x1)
Mon Nov 28 00:51:54 2022 user.emerg syslog: 	/go/src/github.com/lsochanowski/GoHeishaMon/taw2mqtt.go:1060 +0x94
Mon Nov 28 00:51:54 2022 user.emerg syslog: main.main()
Mon Nov 28 00:51:54 2022 user.emerg syslog: 	/go/src/github.com/lsochanowski/GoHeishaMon/taw2mqtt.go:572 +0xa80

```

burt2902, 2022-11-28:
GoHeishaMonConfig.new used:Readonly=true
```
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID=""
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.1.2"
MqttPort="1883"
MqttLogin="aquarea"
MqttPass="hasl0Aquar3A"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true
```

burt2902, 2022-11-28:
Entire GoHeishaMonConfig.new used:
```
Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID=""
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.1.2"
MqttPort="1883"
MqttLogin="aquarea"
MqttPass="REMOVED"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true
```

burt2902, 2022-11-29:
LuCi kernel log

```
[    0.000000] Linux version 3.14.0 (root@A2W-LINUX) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.01 r40838) ) #1 Tue Mar 10 07:21:14 UTC 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9533 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803a2420, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=CUS531 console=ttyS0,115200 mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env),6464k(rootfs),6464k(rootfs2),64k(apl),64k(config),1472k(kernel1),1472k(kernel2),64k(art) rootfstype=squashfs,jffs2 noinitrd mem=64M rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 60692K/65536K available (2725K kernel code, 127K rwdata, 580K rodata, 208K init, 185K bss, 4844K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:650.000MHz, DDR:597.387MHz, AHB:216.666MHz, Ref:25.000MHz
[    0.000000] Calibrating delay loop... 432.53 BogoMIPS (lpj=2162688)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] NET: Registered protocol family 16
[    0.080000] MIPS: machine is Qualcomm Atheros CUS531 reference board
[    0.090000] ar71xx: invalid MDIO id 1
[    0.090000] ar71xx: using random MAC address for eth0
[    0.500000] bio: create slab <bio-0> at 0
[    0.510000] Switched to clocksource MIPS
[    0.520000] NET: Registered protocol family 2
[    0.520000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.520000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.530000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.540000] TCP: reno registered
[    0.540000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.540000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.550000] NET: Registered protocol family 1
[    0.550000] PCI: CLS 0 bytes, default 32
[    0.560000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.580000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.580000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.590000] msgmni has been set to 118
[    0.600000] io scheduler noop registered
[    0.600000] io scheduler deadline registered (default)
[    0.610000] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.630000] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 1562500) is a 16550A
[    0.640000] console [ttyS0] enabled
[    0.650000] bootconsole [early0] disabled
[    0.660000] m25p80 spi0.0: found gd25q128, expected m25p80
[    0.670000] m25p80 spi0.0: gd25q128 (16384 Kbytes)
[    0.670000] 9 cmdlinepart partitions found on MTD device spi0.0
[    0.680000] Creating 9 MTD partitions on "spi0.0":
[... 80 lines cut ...]
[   10.320000] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   10.330000] ieee80211 phy0: Atheros AR9531 Rev:2 mem=0xb8100000, irq=47
[   10.340000] cfg80211: Calling CRDA for country: DE
[   10.340000] cfg80211: Regulatory domain changed to country: DE
[   10.350000] cfg80211:  DFS Master region: ETSI
[   10.350000] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   10.360000] cfg80211:   (2400000 KHz - 2483000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   10.370000] cfg80211:   (5150000 KHz - 5250000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   10.380000] cfg80211:   (5250000 KHz - 5350000 KHz @ 80000 KHz, 200000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   10.390000] cfg80211:   (5470000 KHz - 5590000 KHz @ 80000 KHz, 255000 KHz AUTO), (N/A, 2700 mBm), (0 s)
[   10.400000] cfg80211:   (5590000 KHz - 5650000 KHz @ 40000 KHz, 255000 KHz AUTO), (N/A, 2700 mBm), (600 s)
[   10.410000] cfg80211:   (5650000 KHz - 5725000 KHz @ 40000 KHz, 255000 KHz AUTO), (N/A, 2700 mBm), (0 s)
[... cut ...]
[   22.230000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   22.260000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   22.810000] eth1: link up (1000Mbps/Full duplex)
[   22.810000] br-lan: port 1(eth1) entered forwarding state
[   22.820000] br-lan: port 1(eth1) entered forwarding state
[   22.820000] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   22.860000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.820000] br-lan: port 1(eth1) entered forwarding state
```

wolnypl, 2022-12-04:
Hello, 
First of all, @lsochanowski thanks for your  project.  I'm interested in it a lot. 

I have exactly the same error message o newly-bought cz-taw1  as burt2902
 
I'm also interested with answer to above questions or any other suggestions how this could be fixed. 
===
Sun Dec  4 14:44:56 2022 user.emerg syslog: panic: runtime error: invalid memory address or nil pointer dereference
Sun Dec  4 14:44:56 2022 user.emerg syslog: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x303c3c]
Sun Dec  4 14:44:56 2022 user.emerg syslog: 
Sun Dec  4 14:44:56 2022 user.emerg syslog: goroutine 1 [running]:
Sun Dec  4 14:44:56 2022 user.emerg syslog: go.bug.st/serial.(*unixPort).Write(0x0, 0x5d4a40, 0x6e, 0x6e, 0x1223ac00, 0x8b, 0xb2c97000)
Sun Dec  4 14:44:56 2022 user.emerg syslog: 	/go/src/go.bug.st/serial/serial_unix.go:88 +0x30
Sun Dec  4 14:44:56 2022 user.emerg syslog: main.send_command(0x5d4a40, 0x6e, 0x6e, 0x6e, 0x1)
Sun Dec  4 14:44:56 2022 user.emerg syslog: 	/go/src/github.com/lsochanowski/GoHeishaMon/taw2mqtt.go:1060 +0x94
Sun Dec  4 14:44:56 2022 user.emerg syslog: main.main()
Sun Dec  4 14:44:56 2022 user.emerg syslog: 	/go/src/github.com/lsochanowski/GoHeishaMon/taw2mqtt.go:572 +0xa80
Sun Dec  4 14:44:58 2022 user.emerg syslog: - init complete -

===

sauliuslt, 2022-12-24:
Any progress on this issue? New cz-taw1 devices do not work with current version.

lsochanowski, 2022-12-24:
Hello i've ocerlooked that now i'am processing the issue ;)

sauliuslt, 2022-12-24:
> Hello i've ocerlooked that now i'am processing the issue ;)

Thanks! I hope this is easy fix.

lsochanowski, 2022-12-24:
ok like you all say , we need to upgrade the core system , (openwrt) , i don't have new system , we need to extract it in the way we do it before (by locing upgrade mechanism and let orginal taw to download file but not to install ) when we get it we can replace wrt here in repo

lsochanowski, 2022-12-24:
https://github.com/lsochanowski/GoHeishaMon/tree/NoWebSerwerButtonsAndLED/OS < here is whole openwrt we use today  problem is i don't have the pump or taw anymore we need someone with that hardware to process the case

sauliuslt, 2022-12-24:
How can we help? I have new cz-taw1 with pump

MiG-41, 2022-12-24:
@lsochanowski ,@sauliuslt , here is kernel :  https://www.mediafire.com/file/5ray9x0d70gjytu/openwrt-ar71xx-generic-cus531-16M-kernel.bin/file  , and filesystem: https://www.mediafire.com/file/gaqaqff8h9u3jak/openwrt-ar71xx-generic-cus531-16M-rootfs-squashfs.bin/file 
Orginal for CZ-TAW1  from September 2022.... 
You can try to use only the kernel from it....

MiG-41, 2022-12-24:
What is showing "cat /sys/kernel/debug/usb/devices" command:? , since at my it is:
`root@OpenWrt:/# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  4/800 us ( 1%), #Int=  2, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 3.14
S:  Manufacturer=Linux 3.14.0 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=ehci-platform
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=058f ProdID=6254 Rev= 1.00
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=067b ProdID=2303 Rev= 4.00
S:  Manufacturer=Prolific Technology Inc.
S:  Product=USB-Serial Controller D
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=pl2303
E:  Ad=81(I) Atr=03(Int.) MxPS=  10 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=83(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
`

sauliuslt, 2022-12-25:
I cannot access device via ssh, only luci web admin is working for me.

MiG-41, 2022-12-25:
Here is the newest orginal fw with a2wmain version 04.01.03 : https://www.mediafire.com/file/9z3ewl5g1aqqgk0/download_file.tgz/file

You can try to flash it , it will be orginal FW , if this will work with Service Cloud (connecting ,ect) , then it is included new drivers ,case will be only to upload GoHeishaMon there instead a2wmain....

sauliuslt, 2022-12-26:
Uploaded this FW and it works perfectly.

MiG-41, 2022-12-26:
You uploaded it with sysupgrade as well (so booth files ) and  Smart/Service Cloud works perfectly , or you upload only Kernel file with GoHeishamon sysupgrade file , and GoHeishamon works perfectly ?:)

sauliuslt, 2022-12-26:
I uploaded both files from your archive file. 
I can try different combination, but you have to tel what exact files to upload.

MiG-41, 2022-12-26:
Probably this not helps.... 
@lsochanowski are you able to complile GoHeishamon taking as tamplate latest orginal files ?
I uplated some of them , but it is not compiling automaticly any more...

lsochanowski, 2022-12-26:
on it @MiG-41

lsochanowski, 2022-12-26:
fixed @MiG-41

MiG-41, 2022-12-26:
@lsochanowski maybe you are ale to update all files from nawet orginal squash file? Since i update only pl2303.ko (hoping that this will be ok,but to be sure would be better to update all...

MiG-41, 2022-12-26:
Anyway i tried newest 1.0.187 from @lsochanowski , and it run's ( so at least it will not stuck). It has changed/updated driver for pl2303 , so there is a chance ,that will ork , so @sauliuslt ,please try it.

sauliuslt, 2022-12-27:
Uploaded latest files, but i don't see any messages. I suspect it did not read config file because after flashing FW white LED was solid and never turned off as in previous versions. I still cannot ssh " Connection refused" so i cannot check any logs.

sauliuslt, 2023-01-02:
Any advice on this issue? I wonder why ssh connection is refused?

burt2902, 2023-01-02:
@sauliuslt, you need to have ssh server enabled but it is disabled until you perform a first boot procedure. Luckily, there is a way to spawn the ssh server via MQTT as I have described [here](https://github.com/lsochanowski/GoHeishaMon/issues/23#issuecomment-1332254250). Make sure goheishamon is running and sending/responding to MQTT commands. Since it was crashing for me due to lack of /dev/ttyUSB0, I had to modify the config file and change the device to /dev/ttyS0 first. Hope this lets you move forward.

MiG-41, 2023-01-08:
Try this one: https://www.mediafire.com/file/c2ac2ls56fhzxbz/GoHeishaMon1.1.191z166.zip/file
It should have new kernel , so hopfully support PL23a3 .

burt2902, 2023-01-09:
Thank you for getting the kernel updated to support PL23a3.

PS: I will wait with closing the issue until there is an official release containing this change. I believe it would also make sense to update README.md to point at a release supporting newest CZ_TAW1 hardware.

MiG-41, 2023-01-09:
@burt2902 so i understand ,that ou test it and it is working On your CZ-TAW1, have mqtt topics ,ect ?

burt2902, 2023-01-09:
Yes, confirming that the issue is fixed with the kernel that properly handles 23a3 USB serial chip

omenukas, 2023-01-21:
> Uploaded latest files, but i don't see any messages. I suspect it did not read config file because after flashing FW white LED was solid and never turned off as in previous versions. I still cannot ssh " Connection refused" so i cannot check any logs.

Same problem with me. Did you solve it?