lsochanowski/GoHeishaMon #27: A lack of  messages on MQTT TOPICS

donpablos, 2023-01-02:
I installed the latest version of GoHeishaMon (1.0.166) to CZ-TAW1. 

1/ There is one message on MQTT’s topic:

      panasonic_heat_pump/commands/LWT=offline

2/ In the Mqtt's log:

      1672592003: New connection from 192.168.1.99:41322 on port 1883.
      1672592003: New client connected from 192.168.1.99:41322 as GoHeishaMon-pub (p2, c1, k60, u'haMqttUser').
      1672592003: Client GoHeishaMon-pub closed its connection.

After three days I still can't see any new data.

Please help :)

CONFIG:

Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID="BXX"
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.1.X"
MqttPort="1883"
MqttLogin="XXX"
MqttPass="XXX"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true

donpablos, 2023-01-02:
On my CZ-TAW1 
POWER light is stable green
STATUS light is stable white.

On USB GOHEISHMON create two empty files

04.01.03_007XXX.txt
007XXX.txt

XXX is my CZ-TAW1's ID.

donpablos, 2023-01-06:
@lsochanowski  Lukasz @MiG-41 Marcin pomożecie ? :)

lsochanowski, 2023-01-06:
pomozemy ale to dziwna sprawa goheisha nie tworzy zadnych plikow

MiG-41, 2023-01-07:
> On my CZ-TAW1 POWER light is stable green STATUS light is stable white.
> 
> On USB GOHEISHMON create two empty files
> 
> ```
> 04.01.03_007XXX.txt
> 007XXX.txt
> 
> XXX is my CZ-TAW1's ID.
> ```

Files are made by orginal FW just before update , it is normal.

Try this GoHeishaMon , i is with new kernel but with old GoHeishaMon app , it's working on my CZ-TAW1 (but it have old PL2303 chip). https://www.mediafire.com/file/c2ac2ls56fhzxbz/GoHeishaMon1.1.191z166.zip/file

donpablos, 2023-01-08:
Marcin, Łukasz - you hit the jackpot ! :)  Now it's working perfect. Big thanks !!!!

ciapek33, 2023-03-26:
> pomozemy ale to dziwna sprawa goheisha nie tworzy zadnych plikow

Witam, dali byście radę opisać mi jak uruchomić ten dodatek ? Wgrałem na pena te pliki z ZIP któe tu podaliście, podpinam do CZ-TAW1 i wciskam trzy przyciski i czekam około 10s aż zacznie wgrywać nowy firmware. A co po stronie HA mam zrobić ?

dimaregabriele, 2023-03-30:
can you help me ? I installed the software on the CZ-TAW1. Everything seems ok, (steady green light, steady white light) but on home assistant I can't configure MQTT I don't know where I'm wrong. Is the configuration file like this correct?

`Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID="B7261-31700"
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="192.168.1.37"
MqttPort="1883"
MqttLogin="aa"
MqttPass="aa"
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true`

[image]