lsochanowski/GoHeishaMon #29: CZ-TAW1 crashed after the original firmware update

omenukas, 2023-01-21:
The CZ-TAW1 module was connected to the Aquarea Smart Cloud and stopped working sometime in the fall/winter. The "Status" indicator started flashing red. After rebooting the module it works for a while and then breaks again.
I decided to try sending the heat pump data to the Mqtt server.
I performed a firmware rewrite using the following link:
https://www.mediafire.com/file/c2ac2ls56fhzxbz/GoHeishaMon1.1.191z166.zip/file
Everything started working and I was already happy, but only half an hour. And then the data to mqtt stopped going. After rebooting the module, it's the same again - it works for a while and then stops again. The situation is the same as with the original firmware.
Do you have any ideas on how to solve this problem?

omenukas, 2023-01-22:
Addition:
I have found that when I stop receiving mqtt messages from the heat pump, I can still send setting changes to the heat pump and they actually change on the unit.

donpablos, 2023-01-31:
I'm using firmware from the link:
https://www.mediafire.com/file/c2ac2ls56fhzxbz/GoHeishaMon1.1.191z166.zip/file

I've the same problems like @omenukas

donpablos, 2023-01-31:
I tried the procedure   to access ssh  ( to gather any  logs) :

  _panasonic_heat_pump/commands/OSCommand  umount /overlay
  panasonic_heat_pump/commands/OSCommand jffs2reset -y
  panasonic_heat_pump/commands/OSCommand  reboot_

and after rebooting your software  stops working, only helps me flash the device again .

dimaregabriele, 2023-03-24:
hello, were you able to configure homeassistant ? you can help me ? @donpablos  @omenukas

tomeeka, 2023-04-17:
@omenukas 
I have the same problem with the original firmware, it works for about a day.
After a reboot, it is working again for a while.
Have not tried GoHeishaMon yet.
Do you have any solution to solve this problem? Can this be a HW problem?
My adapter is brand new, doing this since day 1.

omenukas, 2023-04-18:
@dimaregabriele 
Try this perfect tool:
https://github.com/edterbak/NodeRed_Heishamon_control

adnovea, 2023-11-05:
**GoHeishaMon crashes randomly**
I wrote a short script to run GHM and check when it crashes.
Actually after few minutes to several hours.
```
Start GoHeishaMon
-----------------
Sun Nov  5 08:21:51 CET 2023 Start GoHeishaMon Process
Sun Nov  5 08:44:11 CET 2023 Crash ...
Sun Nov  5 08:44:12 CET 2023 Start GoHeishaMon Process
Sun Nov  5 08:54:06 CET 2023 Crash ...
Sun Nov  5 08:54:07 CET 2023 Start GoHeishaMon Process
Sun Nov  5 09:27:14 CET 2023 Crash ...
Sun Nov  5 09:27:15 CET 2023 Start GoHeishaMon Process
Sun Nov  5 09:41:03 CET 2023 Crash ...
Sun Nov  5 09:41:04 CET 2023 Start GoHeishaMon Process
Sun Nov  5 09:49:42 CET 2023 Crash ...
Sun Nov  5 09:49:43 CET 2023 Start GoHeishaMon Process
Sun Nov  5 10:54:29 CET 2023 Crash ...
Sun Nov  5 10:54:30 CET 2023 Start GoHeishaMon Process
Sun Nov  5 11:12:05 CET 2023 Crash ...
Sun Nov  5 11:12:06 CET 2023 Start GoHeishaMon Process
Sun Nov  5 13:05:24 CET 2023 Crash ...
Sun Nov  5 13:05:25 CET 2023 Start GoHeishaMon Process
```
I don't have the knowledge to dig deep into the code and find out what's going on, so I wrote a workaround daemon script file to overcome such a problem.

The BASH script (_see this [post ](https://github.com/lsochanowski/GoHeishaMon/issues/23#issuecomment-1793794214)about how to copy files if needed_) is below : 
```
#!/bin/sh
 [ -z `pidof GoHeishaMon_MIPSUPX` ] || kill -9 `pidof GoHeishaMon_MIPSUPX`
echo -e "\nStart GoHeishaMon\n-----------------\n"
sleep 1
while true
do
echo -e "`date` Start GoHeishaMon Process"
/usr/bin/GoHeishaMon_MIPSUPX | grep -q "out of time for read" || kill -9 `pidof GoHeishaMon_MIPSUPX`
echo -e "`date` Crash ...\n"
sleep 1
done
```

**Launch script at startup**
in LuCi  _System > Startup_ (_bottom of the page_) _Local Startup_ and in the text comment and add the following lines : 
```
# /usr/bin/GoHeishaMon_MIPSUPX > /dev/ttyS0
/etc/gh/daemon.sh > /dev/ttyS0
```

**Conclusions**
Certainly not the best solution because it does not fix the problem. For me it works and enable to read and write parameter to my Aquarea. Any improvements or alternative solutions are welcomed.