lsochanowski/GoHeishaMon #30: CZ-TAW1B compatible

dimaregabriele, 2023-03-08:
I have the latest model, the CZ-TAW1B, does anyone know if it's compatible?

MiG-41, 2023-03-08:
Do you have a pictures ? Best from PCB from booth sides...
Put some pendrive and press 3 buttons at the same time for 10s with changing led colors ?

dimaregabriele, 2023-03-09:
A pendrive empty ?

dimaregabriele, 2023-03-09:
@MiG-41 
[image]
[image]

MiG-41, 2023-03-12:
Looks identical to my board , so should be fine.
There is even PL2303 chip , but expected new PL23A3 like here: https://github.com/lsochanowski/GoHeishaMon/issues/26

Shoud be fine.

dimaregabriele, 2023-03-13:
OK I try. What guide can I follow step by step? @MiG-41

MiG-41, 2023-03-13:
Try with this: https://github.com/lsochanowski/GoHeishaMon/issues/26#issuecomment-1374770882

dimaregabriele, 2023-03-16:
@MiG-41 do i just have to upload the two files on the usb stick? then it plugs into the CZ-TAW1b and I press the 3 buttons at the same time ?

MiG-41, 2023-03-16:
Yes and no :) 
You have to create also third file with configuration.

dimaregabriele, 2023-03-16:
`Readonly=true
Device="/dev/ttyUSB0"
Loghex=true
ReadInterval=1
Aquarea2mqttCompatible=false
Aquarea2mqttPumpID=""
Mqtt_topic_base="panasonic_heat_pump/main"
Mqtt_set_base="panasonic_heat_pump/commands"
ForceRefreshTime=300
MqttServer="127.0.0.1"
MqttPort="1883"
MqttLogin=""
MqttPass=""
SleepAfterCommand=1
MqttClientID="GoHeishaMon-pub"
MqttKeepalive=60
EnableCommand=true
HAAutoDiscover=true`

 this right? and how should i configure for home assistant ?

MiG-41, 2023-03-16:
File should have name GoHeishaMonConfig.new
All you have in Readme on first page of that git: https://github.com/lsochanowski/GoHeishaMon
Section Install...

dimaregabriele, 2023-03-17:
ok i think i managed to install the software. now the device has the power light on green and the status light on white. The link is disabled. it is connected via LAN cable. now what should I do to see on home assistant ? @MiG-41

MiG-41, 2023-03-17:
Nothing...
First you should see topics in MQTT ( ou can use i.e. MQTT Explorer to check it).
In HA you have to make your own integration ,or use template like this: https://github.com/Egyras/HeishaMon/tree/master/Integrations/Home%20Assistant

dimaregabriele, 2023-03-19:
i put the heishamon.yaml file, but i see all the sensors as "unknown" status.
How can I verify that MQTT is working?
In the configuration file I put MqttServer="192.168.1.37", where 192.168.1.37 is the ip address of my homeassistant. it's correct? @MiG-41

MiG-41, 2023-03-20:
> How can I verify that MQTT is working?

Answer already was --> 

> ou can use i.e. MQTT Explorer to check it

dimaregabriele, 2023-03-20:
this is what i see on mqtt explorer.

[image]

MiG-41, 2023-03-30:
Looks like GoHeishaMon is not conected to MQTT ( otherwise you should see topic "panasonic_heat_pump" next to "$SYS").
please check , if there is correct name for config file , and if the config file it self is ok.

dimaregabriele, 2023-03-31:
they are correct? @MiG-41 

[image]
[image]

dimaregabriele, 2023-04-01:
ok i reloaded the program and it worked. For a few hours instead it hasn't communicated anymore and this one comes out on MQTT. what does it mean? @MiG-41 

[image]

MiG-41, 2023-04-01:
It looks like GoHeishaMon process stopped... It is not only you , who complain about...
Unfortunatly @lsochanowski have no more Aquarea HP , and developing of this project is hard...

dimaregabriele, 2023-04-03:
if I enter the device via browser (openwrt) and reboot the system then it starts again. what can it be ? actually I had problems even with the original software. @MiG-41

pando85, 2023-10-16:
I just tested it today and it worked without any problem.

This is the summary of the instructions that I followed:

# cz-taw1b

Panasonic CZ-TAW1B WiFi adapter for Panasonic AQUAREA.

## WiFi configuration

WiFi should be configured in original firmware.

### Setting up WiFi Without WPS on CZ-TAW1

In the paper instructions that come with the CZ-TAW1, there's no mention of setting up WiFi without
WPS. However, in the PDF instructions found on the CD-ROM included with the device and various
online manuals, you'll find a procedure for configuring WiFi settings without using WPS.

1. **Using the HTML Utility**: The CD-ROM contains a small HTML utility that simplifies the process
   of configuring WiFi settings. This utility allows you to enter your WiFi SSID and password, which
   it then saves in a `settings.txt` file for you.

2. **USB Drive Preparation**: To proceed, insert a USB drive into your computer. We recommend using
   an 8GB FAT32-formatted USB drive (although your mileage may vary with other configurations).

3. **Create `settings.txt`**: In the utility, you'll specify your WiFi settings. The `settings.txt`
   file should have the following content (without the quotes, and there should be a newline after
   each key):

   ```plaintext
   SSID=YourSSIDHere
   KEY=APasswordBetterThanThis
   ```

4. **Transfer `settings.txt`**: Save the `settings.txt` file to the root directory of your USB
   drive.

5. **WiFi Configuration**: With the `settings.txt` file on the USB drive, insert it into the
   CZ-TAW1.

6. **Configure WiFi**: To configure the WiFi settings, press and hold the WPS button on the CZ-TAW1
   for 10 seconds. The device will read the `settings.txt` file and set up the WiFi accordingly.

7. **Final Steps**: Once the WiFi configuration is complete, you can remove the USB drive and
   install the transmitter wherever you prefer.

These steps allow you to set up your WiFi on the CZ-TAW1 without the need for WPS.

## Install instructions

New hardware should use 1.0.191 to avoid problems with PL23a3 drivers.
[Original link](https://github.com/lsochanowski/GoHeishaMon/issues/26#issuecomment-1374770882)

To install the software, follow these steps:

1. Format a USB drive to FAT32 and copy the following files to it:

   - `openwrt-ar71xx-generic-cus531-16M-kernel.bin`
   - `openwrt-ar71xx-generic-cus531-16M-rootfs-squashfs.bin`

2. Additionally, configure and copy the file named `GoHeishaMonConfig.new`.

3. Insert the USB drive with these files into your CZ-TAW1 device.

4. Press all three buttons simultaneously and hold them for more than 10 seconds. Wait until the
   middle LED on the CZ-TAW1 begins changing colors, cycling through green, blue, and red. You may
   also notice the LED on the USB drive blinking if it has one.

5. The update process will start, and it will take approximately 3 minutes. During this time, the
   CZ-TAW1 will reboot. After a while, you will see the middle LED light up in white.

6. Do not remove the drive from the module until the white LED turns off again. This indicates that
   the GoHeishaMon has copied the config file from the drive and rebooted the CZ-TAW1. Remove the
   drive before the white LED turns on again, as leaving the drive with the config file present will
   result in it being copied again and triggering another reboot.

### Commands

```bash
sudo mount /dev/sdb1 /mnt
sudo cp 1.0.191/openwrt-ar71xx-generic-cus531-16M-* /mnt
sudo cp GoHeishaMonConfig.new /mnt
sudo umount /mnt
```

## Configuration

### SSH Connection

ssh was not working and dropbear doesn't started automatically. The solution was to start it through
MQTT messages. Topic for sending messages: `panasonic_heat_pump/commands/OSCommand` Topic for
reading output: `panasonic_heat_pump/commands/OSCommand/out`

Just send a `/usr/sbin/dropbear`. Example:

```bash
mosquitto_pub -t "panasonic_heat_pump/commands/OSCommand" -m "/usr/sbin/dropbear" -h <MQTT BROKER IP>
```

For connecting to ssh weaker algorithms are needed:

```bash
ssh -oHostkeyAlgorithms=+ssh-rsa -oKexAlgorithms=+diffie-hellman-group1-sha1 root@${PANASONIC_IP}
```

### Changing hostname

`/etc/config/system`

```bash
uci set system.@system[0].hostname='cz-taw1b'
uci commit system
/etc/init.d/system reload
```

### Fix dropbear

```bash
root@cz-taw1b:~# cat /etc/config/dropbear
config dropbear
    option PasswordAuth 'on'
    option RootPasswordAuth 'on'
    option Port            '22'
#    option BannerFile    '/etc/banner'
```

Set `~/.ssh/config`:

```conf
Host cz-taw1b.iot.grigri
    User root
    PubkeyAcceptedKeyTypes +ssh-rsa
    HostkeyAlgorithms +ssh-rsa
    KexAlgorithms +diffie-hellman-group1-sha1
```

Add rsa key to `/etc/dropbear/authorized_keys` or use LuCi web UI.

### Configure NTP

Change NTP servers to your preferred ones.

dimaregabriele, 2023-11-09:
what command should be sent to start mqtt (from mqtt loupe app)? @pando85

volser, 2024-11-25:
Where I can find original CD-ROM files? what keys except `SSID` and `KEY` settings file supports?

Fedooo2, 2025-11-22:
Any idea where I can find the latest firmware for CZ-TAW1B? I have problems connecting it to network and the  tech guy told me that if I have an older module that was not connected for some time then it might not be possible anymore due to outdated firmware. Therefore I would like to chip it with the latest firmware and try.

Fedooo2, 2025-11-23:
Great, thanks a lot. It is definitely better than mine firmware. I guess I just need to copy both files to a FAT32-formatted USB drive, insert the  drive into the  device  and press all three buttons simultaneously and hold them for more than 10 seconds. I will give it a try when I will be at my mother in law where the pump is installed.