lsochanowski/GoHeishaMon #33: CZ-TAW1 reflash

pampyras, 2023-06-15:
Little offtopic, but:
Does anyone know how to reflash to and/or has original firmware?
Device is in infinite reboot loop, so cannot be flashed with usb drive, only by serial with tftpd.
Tried different kernel versions, same.

MiG-41, 2023-06-16:
Depends what you did....
Try to change side , from printenv you will be able to see ,on with side you are actually:
https://github.com/Egyras/HeishaMon/issues/19#issuecomment-630376848

pampyras, 2023-06-16:
Thanks, it looks like side 1.
Maybe you can share address of rootfs to copy to?

MiG-41, 2023-06-16:
This is valid for SIDE 1 ( when is SIDE 2  mtd2 is swapped with mtd4 , mtd7 with mtd8 - or something like that)

[    0.670000] 9 cmdlinepart partitions found on MTD device spi0.0
[    0.680000] Creating 9 MTD partitions on "spi0.0":
[    0.680000] 0x000000000000-0x000000040000 : "u-boot"
[    0.690000] 0x000000040000-0x000000050000 : "u-boot-env"
[    0.690000] 0x000000050000-0x0000006a0000 : "rootfs"
[    0.700000] mtd: device 2 (rootfs) set to be root filesystem
[    0.710000] 1 squashfs-split partitions found on MTD device rootfs
[    0.710000] 0x000000460000-0x0000006a0000 : "rootfs_data"
[    0.720000] 0x0000006a0000-0x000000cf0000 : "rootfs2"
[    0.730000] 0x000000cf0000-0x000000d00000 : "apl"
[    0.730000] 0x000000d00000-0x000000d10000 : "config"
[    0.740000] 0x000000d10000-0x000000e80000 : "kernel1"
[    0.750000] 0x000000e80000-0x000000ff0000 : "kernel2"
[    0.750000] 0x000000ff0000-0x000001000000 : "art"
[    0.780000] libphy: ag71xx_mdio: probed
[    1.410000] ag71xx ag71xx.0: connected to PHY at ag71xx-mdio.1:04 [uid=004dd042, driver=Generic PHY]
[    1.420000] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode:MII

root@OpenWrt:/# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00650000 00010000 "rootfs"
mtd3: 00240000 00010000 "rootfs_data"
mtd4: 00650000 00010000 "rootfs2"
mtd5: 00010000 00010000 "apl"
mtd6: 00010000 00010000 "config"
mtd7: 00170000 00010000 "kernel1"
mtd8: 00170000 00010000 "kernel2"
mtd9: 00010000 00010000 "art"

pampyras, 2023-06-16:
I tried to boot from kernel1 and kernel2, same boot loop. Maybe propblem with hardware? Before kernel update I was getting squashfs coruption errors.

`U-Boot 1.1.4-g4d980152-dirty (Jun 24 2016 - 10:12:23)

cus531 - Honey Bee 2.0DRAM:
sri
Honey Bee 2.0
ath_ddr_initial_config(195): (16bit) ddr2 init
tap = 0x00000003
Tap (low, high) = (0x7, 0x20)
Tap values = (0x13, 0x13, 0x13, 0x13)
setup gpio
64 MB
Top of RAM usable for U-Boot at: 84000000
Reserving 199k for U-Boot at: 83fcc000
Reserving 192k for malloc() at: 83f9c000
Reserving 44 Bytes for Board Info at: 83f9bfd4
Reserving 36 Bytes for Global Data at: 83f9bfb0
Reserving 128k for boot params() at: 83f7bfb0
Stack Pointer at: 83f7bf98
Now running in RAM - U-Boot at: 83fcc000
Flash Manuf Id 0xc8, DeviceId0 0x40, DeviceId1 0x18
flash size 16MB, sector count = 256
Flash: 16 MB
*** Warning *** : PCIe WLAN Module not found !!!
In:    serial
Out:   serial
Err:   serial
Net:   ath_gmac_enet_initialize...
No valid address in Flash. Using fixed address
No valid address in Flash. Using fixed address
ath_gmac_enet_initialize: reset mask:c02200
Scorpion ---->S27 PHY*
S27 reg init
: cfg1 0x800c0000 cfg2 0x7114
eth0: 00:03:7f:09:0b:ad
athrs27_phy_setup ATHR_PHY_CONTROL 4 :1000
athrs27_phy_setup ATHR_PHY_SPEC_STAUS 4 :10
eth0 up
Honey Bee ---->  MAC 1 S27 PHY *
S27 reg init
ATHRS27: resetting s27
ATHRS27: s27 reset done
: cfg1 0x800c0000 cfg2 0x7214
eth1: 00:03:7f:09:0b:ad
athrs27_phy_setup ATHR_PHY_CONTROL 0 :1000
athrs27_phy_setup ATHR_PHY_SPEC_STAUS 0 :10
athrs27_phy_setup ATHR_PHY_CONTROL 1 :1000
athrs27_phy_setup ATHR_PHY_SPEC_STAUS 1 :10
athrs27_phy_setup ATHR_PHY_CONTROL 2 :1000
athrs27_phy_setup ATHR_PHY_SPEC_STAUS 2 :10
athrs27_phy_setup ATHR_PHY_CONTROL 3 :1000
athrs27_phy_setup ATHR_PHY_SPEC_STAUS 3 :10
eth1 up
eth0, eth1
Setting 0x181162c0 to 0x4891a100
Hit any key to stop autoboot:  0
## Booting image at 9fd10000 ...
   Image Name:   MIPS OpenWrt Linux-3.14
   Created:      2022-11-21   5:39:46 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1223035 Bytes =  1.2 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum at 0x9fd10040 ...OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80060000) ...
## Giving linux memsize in bytes, 67108864

[... 81 lines cut ...]
[    3.030000] 8021q: 802.1Q VLAN Support v1.8
[    3.040000] VFS: Mounted root (squashfs filesystem) readonly on device 31:2.
[    3.050000] Freeing unused kernel memory: 208K (803bc000 - 803f0000)
procd: Console is alive
procd: - watchdog -
procd: - preinit -
[    4.880000] random: mktemp urandom read with 41 bits of entropy available
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
 [    5.970000] usbcore: registered new interface driver usbfs
[    5.980000] usbcore: registered new interface driver hub
[    5.980000] usbcore: registered new device driver usb
[    5.990000] Button Hotplug driver version 0.4.1
[    6.010000] SCSI subsystem initialized
[    6.020000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    6.030000] ehci-platform: EHCI generic platform driver
[    6.030000] ehci-platform ehci-platform: EHCI Host Controller
[    6.040000] ehci-platform ehci-platform: new USB bus registered, assigned bus number 1
[    6.050000] ehci-platform ehci-platform: irq 3, io mem 0x1b000000
[    6.070000] ehci-platform ehci-platform: USB 2.0 started, EHCI 1.00
[    6.070000] hub 1-0:1.0: USB hub found
[    6.080000] hub 1-0:1.0: 1 port detected

U-Boot 1.1.4-g4d980152-dirty (Jun 24 2016 - 10:12:23)
`

MiG-41, 2023-06-16:
Looks like it is stucking on USB detection , perhaps you have somthing in USB ? Or controller USB-TTL controller is damaged....
Did you set to SIDE 2 as well ?

pampyras, 2023-06-16:
USB is empty. Yes the same problem with side2, reboots at the same place.

MiG-41, 2023-06-16:
This are next steps from log:

[    6.010000] ehci-platform: EHCI generic platform driver
[    6.020000] ehci-platform ehci-platform: EHCI Host Controller
[    6.030000] ehci-platform ehci-platform: new USB bus registered, assigned bus number 1
[    6.040000] ehci-platform ehci-platform: irq 3, io mem 0x1b000000
[    6.060000] ehci-platform ehci-platform: USB 2.0 started, EHCI 1.00
[    6.060000] hub 1-0:1.0: USB hub found
[    6.070000] hub 1-0:1.0: 1 port detected
[    6.080000] usbcore: registered new interface driver usb-storage
[    6.390000] usb 1-1: new high-speed USB device number 2 using ehci-platform
[    6.540000] hub 1-1:1.0: USB hub found
[    6.540000] hub 1-1:1.0: 4 ports detected
^[[    6.830000] usb 1-1.2: new full-speed USB device number 3 using ehci-platform
jffs2 is ready
[    8.210000] jffs2: notice: (341) jffs2_build_xattr_subsystem: complete building xattr subsystem, 1 of xdatum (1 unchecked, 0 orphan) and 19 of xref (0 dead, 0 orphan) found.
switching to jffs2
procd: - early -
procd: - watchdog -
procd: - ubus -
procd: - init -
Please press Enter to activate this console.
[   10.670000] NET: Registered protocol family 10
[   10.700000] nf_conntrack version 0.5.0 (951 buckets, 3804 max)
[   10.710000] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   10.730000] Loading modules backported from Linux version v3.2-rc1-229534-gb3e71d7
[   10.740000] Backport generated by backports.git backports-20150129-0-gdd4a670
[   10.750000] ip_tables: (C) 2000-2006 Netfilter Core Team
[   10.770000] nf_conntrack_rtsp v0.6.21 loading
[   10.790000] nf_nat_rtsp v0.6.21 loading
[   10.820000] usbcore: registered new interface driver usbserial
[   10.820000] usbcore: registered new interface driver usbserial_generic
[   10.840000] usbserial: USB Serial support registered for generic
[   10.890000] xt_time: kernel timezone is -0000

pampyras, 2023-06-16:
Maybe possible to disable usb at all if problem is hardware?

MiG-41, 2023-06-16:
But to do it you should remove usb drivers from kernel.... I don't hink there is a possibility to disable USB from uBoot.