lsochanowski/GoHeishaMon #34: MQTT don't start

yugo000, 2023-10-09:
I installed GoHeishaMon on my CZ-TAW1 dewice but after installation white diode still light.
I have connection via http. Internet works. I connected to tx/rx ports and checked everything. USB works. I have coppied config file to: /etc/gh/config.
But dont see any mqqt traffic.
Strange thing, but dropbear can't start automaticaly. I must start it manually /usr/sbin/dropbear.
I tried to start GoHeishaMon_MIPSUPX manually - dont work.
I checked 1.0.166 version and the latest.
Any ideas?

yugo000, 2023-10-24:
Now I installed newest version.
But stil nothing.
top shows that process GoHeishaMon_MIPSUPX uses about 90% of cpu.
Don't see any trying of connection to my mqtt server.

dimaregabriele, 2023-11-09:
same problem

vytasmk, 2023-12-04:
Similar problem here with versions 1.0.191 and 1.0.194. Config file is not read from USB flash drive. GUI is accessed and looking at running processes `/usr/bin/GoHeishaMon_MIPSUPX` eats 100% CPU and 12% RAM. 

I checked my board it has PL2303 USB to UART chip. PCB date looks like 22 week of 2022

vytasmk, 2023-12-04:
OK I figured out that version mentioned in [here](https://github.com/lsochanowski/GoHeishaMon/issues/27#issuecomment-1374619385) and FW from mediafire link in that comment is working fine for me.