lsochanowski/GoHeishaMon #43: Changing MQTT Broker IP address

mariusz-zapala, 2024-10-30:
Hi

I have installed GoHeishaMon on CZ-TAW1 and it is working great for one year now.
I'm using external MQTT Broker running on IP: 10.0.0.210, but I want to stop using that Broker and move to MQTT addon in Home Assistant, which is IP of 10.0.0.50. How can I change the IP address of MQTT Broker in GoHeishaMon without the need to reinstall everything? I remember installation was not that easy for me and I'm afraid to break it.
Is there an easy/quick way to change MQTT Broker IP?
I tried connecting with ssh, but it asks me for root password and I have no idea what that would be. Can I change MQTT address by ssh? What is root password?

Anyone can help?

Best regards,
Mariusz

MiG-41, 2024-10-31:
As i recall , on each boot GoHeishaMon is looking for file GoHeishaMonConfig.new on USB pendrive , if it is exist ,it get new config from it.
So you need to prepare new GoHeishaMonConfig.new file on USB stick , plug it to CZ-TAW1 and do reboot , and obserwe Pendrive and CZ-TAW1 LED's. If he get's new config it will reboot , so you need to take out pendrive in proper moment ( just after reboot) , otherwise It will be rebooting continiously...

mariusz-zapala, 2024-11-01:
Thanks for quick answer. I'm afraid to break something (I don't have my current configuration file and I don't know how to retrieve it via ssh), so I've decided to let go for now.